openenergymonitor/EmonLib #16: ESP8266 integration/compatibility

giacomoleonzi, 2015-07-15:
Hi
I try to use your library on ESP8266 and I get stuck when the library try to calculate the sqrt() function. The sqrt function lead ESP to crash and reboot, because of some problem in its math library  when you program via the ArduinoIDE.
to overcame the issue, I implemented an approximated method to calculate the squareroot and it works fine.

Do you want that I do a pull request to integrate that part? 
thanks 
regards
Giacomo

glynhudson, 2015-07-16:
Hi Giacomo,

That's great, I would love to try using emonLib with the ESP8266. I have
recently got hold of one. Do you think a pull request on a separate branch
for ESP8266 or starting a totally seperate emonLib_ESP8266 library could be
best? What do you think? I think the latter would be best. Are you using
Arduino IDE to program the ESP? I would be very interested to hear more
about your project. Do you have a blog? If would be great to do a guest
blog on the OpenEnergyMonitior blog regarding using emonLib on ESP8266. I
think many people will be interested in using this new chip

giacomoleonzi, 2015-07-16:
Hi 
Yes I had use the ArduinoIDE to program the ESP.
I think that the solution of separete branch  is enough, because I think that this bug will be fixed.
I hope that you can hear about the project soon, but I don't have a blog right now.
I'm pretty sure about the interest in this chip, so writing on the OpenEnergyMonitor's blog about that could be very intresting.

glynhudson, 2015-07-18:
Great, please submit a pull request and send info about your project when
you have made some progress. I would love to post a blog about the ESP82
EmonLib on our blog. I have recently got myself an ESP, I hope to get going
with it soon. Let me know if you have any advice to get started. Do you
recommend using Arduino IDE to program?

markwalters0, 2016-01-20:
To extend on the above issue with ESP compatibility, it would be good to allow external ADC chips to be used with the library. (eg MCP3008 is a 10 bit 8 channel ADC with SPI, with an existing library MCP3008.h.)
With this we could publish with MQTT straight to emoncms.

mihino, 2016-05-16:
I have just prototyped Arduino Pro Mini with ESP-01. It sends data in 5s avarages via WiFi.
It works stable for a week. Check it: [homecircuits.eu/blog/wifi-energy-monitor/](http://homecircuits.eu/blog/wifi-energy-monitor/)
ESP needs more power for WiFi connection, so it is not suitable for battery supply.

glynhudson, 2016-05-17:
Nice! Good work. We have started work on an emonESP energy monitor also
https://community.openenergymonitor.org/t/emonesp-details/372/3

markwalters0, 2016-06-05:
I have emonTx firmware running directly on the ESP8266 (NodeMCU ESP-12E).
I used a standalone ADC chip (MCP3008, i had some lying around from a RPi build). ESP now has 8 analog inputs
I found the WiFiManager library (https://github.com/tzapu/WiFiManager), which seemed to do everything I needed for the AP setup of the ESP. I did try Glyn's emonESP method first though.
The ESP then sends the results via MQTT to a RPi. 

This is where I am up to at the moment, I either need to work out the MQTT input into emoncms or setup mqttwarn to make a http post input.
I am a complete noob on github, i made my own repo here: 
https://github.com/markwalters0/emonTx_ESP8266

posicat, 2018-03-18:
I read this forum a few days ago, after discovering the ESP8266 only has a single ADC port. Typically I'm not the kind to just take the easy way out, so I looked for a few ADC chips, and found the very nice ADS1115 chips which happily sit on an I2C bus and have 2/4 inputs and 16 Bit resolution.  I ordered some from Adafruit to test how they work, and then wandered over to EasyEDA to design a board for an ESP-01 and 4 ADS1115 chips.

I was looking over the EmonLib code, and it looks like the analog reads are hard-coded, but it seems like those calls could be wrapped in an #ifdef CUSTOM_ADC that would call the native ADC on Arduino's unless it was set, then it would call a custom method like customADCRead, same parameters.  This would let people using alternate ADC chips easily use the EmonLib code.

My project is going to have a few parts, the board is going to live in my laundry/furnace room.  First version will just monitor the washer/dryer and alert me when the wash is done (it's out of the way and hard to hear) eventually I want to expand to reading the blinking LED on my power-meter to monitor whole-house power use, then on to perhaps monitoring the AC, furnace and more.  It'll be a work in progress, but you see why I designed the 16-input board.

https://easyeda.com/posicat/ESP-01-Analog-I2C

vanjsy, 2018-03-19:
HI, is this possible to use same script but using ESP8266? Thanks.

posicat, 2018-03-19:
There will have to be at least a few code changes to support the board I built.  For one, it reads the current through a separate chip.  Once I have a working board at home I can see if I can make a few code modifications for this board.

markwalters0, 2018-03-19:
My version with esp12e and the mcp3008 adc chip works. Reads 8 channels and
sends data by mqtt to rpi running emonhub.

I had to modify the code though and configuring the rpi was tricky (for me
at least).

vanjsy, 2018-03-19:
Is there a way where i will use only ESP8266? Without adc? Current only. Thanks