openenergymonitor/EmonLib #19: RMS computation improvement suggestion

bcaro, 2015-09-13:
I used this very interesting library.

I have some suggestions for calcIrms():
1 - Before entering into the for loop, some variables have to be initialized:
  sampleI = analogRead(inPinI); // very important for a signal with DC component
                                                   // the first sample create a high RMS
  filteredI=0; // not so important but better
  for (int n = 0; n < NUMBER_OF_SAMPLES; n++)

2 - The first order filter is sensitive to sampling period
if the speed of calculation changes (change the frequency of the microcontroller from 2MHz -> 16 MHz) the cutt-off frequency changes also by a factor 8. This is because the sampling rate is not fixed and varies with the computation time. Perhaps its will difficult to impove that, but it is important to know this effect.

3 - The gain of the filter is not equal to 1 at high frequency
but it should be in order to not affect the RMS.
the filter should be equal to:
#define CUTT_OFF 0.996
filteredI =CUTT_OFF*(lastFilteredI+CUTT_OFF*(sampleI-lastSampleI));
With 0.996, the i error is equal to 0.4 %
But if we want a higher cut-off frequency the error increases, because CUTT_OFF is lower.

I hope it will help.

srinivasan251992, 2016-07-22:
what is the significance of using 0.996

Barabba11, 2019-01-20:
Hi, thanks for this library. Can you kindly tell how it works? Will it take the processor busy for some time to "catch" a full sinewave and from it make calculations? Looks to me the only chance. I would like to know if during measurement I can't do something else, and for how much time, thank you.
At least with this library I don't have to add hardware like AC/DC rectifier and it seems already precise..