openenergymonitor/EmonLib #20: Real power

SameerWatve, 2016-01-07:
I am using Arduino mega, acs current sensor, step down transformer with bias (Only positive voltage cycle.)
RMS value of current and voltage is accurate, apparent power also correct.
Real power and thus power factor are inaccurate. There may be a circuit error but deviation is too large.
Please help. Is the code has to be changed for acs current sensor ?

richard-scott, 2016-01-07:
Can you provide a website link to the exact sensor you are using?

SameerWatve, 2016-01-13:
http://www.sunrom.com/p/current-sensor-20a-acs712

richard-scott, 2016-01-18:
How are you connecting this to the AC supply you are measuring?

SameerWatve, 2016-01-19:
Using supply of ac mains.
Supply-acs current sensor-choke-bulb-supply

Step down transformer parallel to load (choke + Bulb)

Aldair170496, 2019-10-30:
Hii! My objective is to read P, S, FP, Vrms and Irms.
My currente calibration is 54.5 using a 33 burden resistor, SCT-013-000.
I am using a 230/9 Vrms transformer, R1=120K e R2=10K, so the voltage calibration i use is 234.26 but the voltage values is stay less than 3V, why?
Help me please!

CODE:
{
Serial.begin(9600);

emon1.voltage(2, 234.26, 1.7); // Voltage: input pin, calibration, phase_shift
emon1.current(1, 54.5); // Current: input pin, calibration (2000/33).
}

void loop()
{
emon1.calcVI(20,2000); // Calculate all. No.of half wavelengths (crossings), time-out
emon1.serialprint(); // Print out all variables (realpower, apparent power, Vrms, Irms, power factor)

float realPower = emon1.realPower; //extract Real Power into variable
float apparentPower = emon1.apparentPower; //extract Apparent Power into variable
float powerFActor = emon1.powerFactor; //extract Power Factor into Variable
float supplyVoltage = emon1.Vrms; //extract Vrms into Variable
float Irms = emon1.Irms; //extract Irms into Variable
delay(1000);
}

Aldair170496, 2019-10-31:
Hi!
I fix the problem, was in the hardware not in the software.
But when the sensor or the voltage transformer are disconnected it should be all zero (0), how to fix that.
Any help would be helpfull.

TrystanLea, 2020-01-30:
Hello @Aldair170496, What kind of values do you see? can you give more detail?