openenergymonitor/EmonLib #21: Emonlib conflicts with other analogRead values.

dhruvvyas90, 2016-01-28:
I've a scenario where I'm using I2C interrupt and Emonlib together. I2C ISR contains several `analogReads` and emonlib measurement is running in `loop()`. I've observed that values from other `analogReads` are not quite correct when I'm using it with emonlib. Upon investigating, I've landed upon this Arduino thread, http://forum.arduino.cc/index.php?topic=54976.0, which suggests that if I'm in a middle of an `analogRead` and try to use the other `analogRead` (probably in an ISR caused by an interrupt) , due to multiplexer switch it might give me a wrong value. Is this a known issue ? If yes, can we enhance `emonlib` to put a certain lock or any other similar mechanism to prevent simultaneous access ? 

Appreciate the help. Thanks.

glynhudson, 2016-01-31:
Interesting issue, not heard the before. Nor am I sure what the fix could
be.

- sent from my mobile device

dhruvvyas90, 2016-01-31:
I can write a small Arduino code to reproduce this issue on Uno if you want me to.

Also I'm comparatively new to Arduino platform but may be making `analogReads` section in Emonlib `atomic` could do the trick to prevent simultaneous access but not sure how it affects reaction time of interrupts afterwords since Arduino documentation (https://www.arduino.cc/en/Reference/AnalogRead) suggests that a single analogRead can take upto 100 us.

dhruvvyas90, 2016-02-19:
Any idea how to solve this conflict ?

richard-scott, 2016-02-19:
Can you upload some code?

If I was writing this, I'd make my main loop do all the analogReads for both my own needs and what emonlib requires and then output the read results into variables that are returned via the ISR. 

I would guess that having an analogeRead's in both an ISR and your main loop is not wise is it? as you never really know when the ISR will trigger and it could overlap your reads in the loop.

...maybe i've miss understood what you are doing so some sample code would help :-)

dhruvvyas90, 2016-02-19:
You've understood it correctly, in fact that's what I'm doing to avoid this issue for now, ie using analog read serially in loop to avoid conflict instead of an ISR . I was just wondering if we can enhance emonlib somehow internally. Like using atomic block or something. Appreciate your help. Thanks.