openenergymonitor/EmonLib #22: Set ADC to 1.1V internal reference?

Harvie, 2016-03-09:
I understood that precision is issue as CT produces very small voltages. Most of arduinos have ability to set ADC to use 1.1V internal reference instead of 5V internal reference. This should make ADC more sensitive (almost 5 times). Can you please add this feature? I'd like to measure current as precisely as possible with such simple circuit. I'd like to have precision down to +-3W (1W would be awesome, 5W max)...

glynhudson, 2016-03-09:
Would this not give us much less range? The voltage divider circuit on the emonTx converts the current output from the CT to a 0>5V voltage output, see https://openenergymonitor.org/emon/buildingblocks/ct-sensors-interface. The ability to measure such low powers will be restricted due to noise, see https://openenergymonitor.org/emon/buildingblocks/measurement-implications-of-adc-resolution-at-low-current-values.

I would recomend you post this issue to the community forums: https://openenergymonitor.org/emon/forum

Harvie, 2016-09-11:
I don't say it should be enabled by default, but it would be cool if this would be option.