openenergymonitor/EmonLib #23: Emonlib with ADS1115

SybrenV, 2016-03-29:
Hello all!

I have a question for you,

I wanted to use the ADS1115 to read the analog value from my diy current clamp pcb(which worked perfectly with a Arduino nano).

Since i am a self taught programmer and i lack skill i cant figure out how I can implement the values I read with the ADS1115 into the emonlib library. Im using the Adafruit library (https://www.adafruit.com/product/1085).

Can maybe somebody point me in the right direction how to do this? I know the analogRead is located in Emonlin.ccp and i tried a lot of variation to load in the right number, but no luck...

Thanks in advance!

glynhudson, 2016-03-29:
Hi, sounds like a great idea. I've not done this myself, just looked at the ADS1115, looks like a tidy package, I'm sure there will be plenty of support out there how to use this. I recomend you post on our forums, failing a good response maybe try the Adafruit forums .

http://openenergymonitor.org/emon/forum

closed 2016-03-29 by glynhudson

fahrvergnuugen, 2018-02-13:
The issue here is that the library expects an Arduino pin in the `current` method but you cannot do it that way if you are using an i2c ADC breakout board. The values from the ADS1115 pins are accessed via methods provided by the Adafruit library, eg `adc0 = ads1015.readADC_SingleEnded(0);`.

It would be nice if emonlib supported analog inputs from the ADS1115 - or at the very least allowed you to pass the analog pin value in the loop via `calcIrms`.

jonathanrjpereira, 2018-04-16:
Is there a reason why you are you using an external ADC such as the ADS1115 instead of the built-in Arduino ADC. Is it simply because the ADS1115 has a higher resolution of 16bits as compared to the 10bit/12bit ADC resolution of the Arduino?

I have worked with the ADS1115 in the past. I wouldn't mind trying to figure something out that will help you.

fahrvergnuugen, 2018-04-16:
Hi Jonathan - it's because I was trying to use emonlib on an ESP8266. The ESP8266 does not have a useable native analog input. (it has one, but trying to rapidly sample it causes the wifi connection to break).

fahrvergnuugen, 2018-04-16:
For what it's worth, I did fork this library and successfully implemented a callback for reading the input that is also backward compatible. I don't know if the author will want to merge my change or not.

https://github.com/fahrvergnuugen/EmonLib/blob/master/examples/current_only/ads1115.ino

posicat, 2018-04-16:
> Hi Guys/Girls,
>
> I did get it to work, by rewriting the emonlib library, but it was
> crontoversly to say at least. But it works perfectly, it still is measuring
> the current.
> The reason I used the ADS1115 was indeed the ESP. I used the Nodemcu board
> and could use the extra current sensors.
>
> Fahrvergnuugen, thanks for posting you file. I will try it out!
>
> Regards.

I did a similar change to get the ADS1115 reading code into Emonlib, but
your solution is more elegant, I like it!

HasseM, 2019-08-18:
> For what it's worth, I did fork this library and successfully implemented a callback for reading the input that is also backward compatible with existing sketches. I don't know if the author will want to merge my change or not.
> 
> https://github.com/fahrvergnuugen/EmonLib/blob/4a965d87061c19ad8b0a35bb173caada014ecbd9/examples/current_only/ads1115.ino#L23

@PaulWieland I tried that code but found some issues. It seems that it does not work with the correct amount of bits...
#if defined(ADS1115_CONVERSIONDELAY)
#define ADC_BITS    15

is simply not working and when forcing it do ADC_BIT 15 all kind of overflows occure (negative power values).

PaulWieland, 2019-08-20:
> > For what it's worth, I did fork this library and successfully implemented a callback for reading the input that is also backward compatible with existing sketches. I don't know if the author will want to merge my change or not.
> > https://github.com/fahrvergnuugen/EmonLib/blob/4a965d87061c19ad8b0a35bb173caada014ecbd9/examples/current_only/ads1115.ino#L23
> 
> @PaulWieland I tried that code but found some issues. It seems that it does not work with the correct amount of bits...
> #if defined(ADS1115_CONVERSIONDELAY)
> #define ADC_BITS 15
> 
> is simply not working and when forcing it to ADC_BIT 15 all kind of overflows occure (negative power values).

I would guess that you have issues with your ADS1115 code and not with emonlib or the fork I made. Try building a separate sketch that just uses the ADS1115 to sample known values. Get that working first and then apply that to the user defined pin reader callback.

felangga, 2019-09-30:
Hei Paul, I'm using your code but it always return 0. Do I need to change anything on ads1115 example code?

korishan, 2020-06-02:
> For what it's worth, I did fork this library and successfully implemented a callback for reading the input that is also backward compatible with existing sketches. I don't know if the author will [...]

Thanks for the changes in updating the code for support of the ADS1115. I am currently working on an ESP32 version and this is quite useful (not able to test atm as I don't have the rest of the parts in yet). I'll give it a shot soon. 
I also added:
"  modified by PaulWieland for ADS1115: https://github.com/PaulWieland/EmonLib/tree/4a965d87061c19ad8b0a35bb173caada014ecbd9"
to the .cpp and .h files to give credit to your changes. 

In my build I'll be using 4 additional modules to be able to scan up to 16 circuits.

Hope it all works flawlessly  :P   But we know in reality it'll kick and have a tantrum.

Kori

Saad-Imtiaz, 2021-09-27:
With 

> Thanks for the changes in updating the code for support of the ADS1115. I am currently working on an ESP32 version and this is quite useful (not able to test atm as I don't have the rest of the [...]

Are you able to use both, the Analog Pins of the ESP32 and as well as the ADS1115 Analog Pins with the current library by  #PaulWieland ?

GianRickes, 2022-02-23:
I did it according to @PaulWieland's code..
https://github.com/fahrvergnuugen/EmonLib/blob/4a965d87061c19ad8b0a35bb173caada014ecbd9/examples/current_only/ads1115.ino#L23
and I used the following libraries...
https://github.com/PaulWieland/EmonLib
https://www.arduino.cc/reference/en/libraries/adafruit-ads1x15/  (2.4.0)
however it presented problems, the result was as the following print of the serial monitor...
[image]