openenergymonitor/EmonLib #29: Error mesage

v3nomers, 2016-09-22:
how to solve it

v3nomers, 2016-09-22:
Arduino: 1.6.8 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Thomas\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Thomas\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Thomas\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10608 -build-path "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Thomas\Documents\Arduino\EmonLib-master\examples\current_only\current_only.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Thomas\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Thomas\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Thomas\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10608 -build-path "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Thomas\Documents\Arduino\EmonLib-master\examples\current_only\current_only.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\variants\standard" "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\variants\standard" "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\variants\standard" "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp" -o "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\variants\standard" "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp" -o "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\variants\standard" "C:\Users\Thomas\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino\wiring_pulse.S" -o "C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\core\wiring_pulse.S.o"
Using previously compiled file: C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\core\WInterrupts.c.o
Using previously compiled file: C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\core\hooks.c.o
Using previously compiled file: C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\core\wiring.c.o
Using previously compiled file: C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\core\wiring_analog.c.o
[... cut ...]
C:\Users\Thomas\Documents\Arduino\EmonLib-master\examples\current_only/current_only.ino:9: undefined reference to `EnergyMonitor::current(unsigned int, double)'

C:\Users\Thomas\AppData\Local\Temp\build03daa4c523108ec5d5526b9f8bce401d.tmp\sketch\current_only.ino.cpp.o: In function `loop':

C:\Users\Thomas\Documents\Arduino\EmonLib-master\examples\current_only/current_only.ino:14: undefined reference to `EnergyMonitor::calcIrms(unsigned int)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

glynhudson, 2016-09-27:
Please post on our community forum. This is not an issue with emonlib

https://community.openenergymonitor.org

- sent from my mobile device

closed 2016-09-27 by glynhudson