openenergymonitor/EmonLib #38: calculations ECS1030-L72 sensor?

AbrahamYu, 2017-05-22:
My Energy Meter parts List
- supply voltage is 5V
- Arduino UNO
- 1 x ECS1030-L72 (DTS : https://dlnmh9ip6v2uc.cloudfront.net/datasheets/Sensors/Current/ECS1030-L72-SPEC.pdf)
- 2 x 10k Ohm resistors 
- 1 x 10uF capacitor
- 1 x Burden resistor 120 Ohms 

[image]

Code
```
#include "EmonLib.h"                   // Include Emon Library
EnergyMonitor emon1;                   // Create an instance

void setup()
{  
  Serial.begin(9600);  
  emon1.current(0, 111.11);             // Current: input pin, calibration.
}

void loop()
{
   
    double Irms = emon1.calcIrms(1480);  
    Serial.print(Irms*220.0);         // Apparent power
    Serial.print(" ");
    Serial.println(Irms, DEC);          // Irms
    delay(1000);
}
```
8.5w lamp measurement

but output data :
137.87  0.6266713142
140.01  0.6364147663
138.00  0.6272707462
137.44  0.6247273445
135.22  0.6146469116
137.73  0.6260547637
139.02  0.6319146156
138.40  0.6290988445
137.97  0.6271401405
139.11  0.6323106288
138.62  0.6300705909
139.56  0.6343578338
141.37  0.6425902843
141.72  0.6441868782
140.05  0.6366049766
139.36  0.6334630012

What's the problem????????

jamipraveeen, 2017-05-22:
>    cloudfront.net/datasheets/Sensors/Current/ECS1030-L72-SPEC.pdf

you are measuring power and current r8!! Inorder to measure power we need
both voltage sensor and current  sensor.As power is product of voltage and
current

AbrahamYu, 2017-05-22:
adding 9v adapter and change code

adapter  
input - AC100~240V 50/60Hz 0.2A
output - DC9V 0.5A

[image]

```
#include "EmonLib.h"             // Include Emon Library
EnergyMonitor emon1;             // Create an instance

void setup()
{  
  Serial.begin(9600);

  emon1.voltage(1, 234.26, 1.7);  // Voltage: input pin, calibration, phase_shift
  emon1.current(0, 111.11);       // Current: input pin, calibration.
}

void loop()
{
  emon1.calcVI(20,2000);         // Calculate all. No.of wavelengths, time-out
  emon1.serialprint();           // Print out all variables
}
```

jonathanrjpereira, 2018-04-16:
In the first image, the ECS1030-L72 has a CT Ratio = 30A/15mA = 2000. You also mentioned that the burden resistance connected is 120ohms.
Hence the CT Sensor Calibration value = CT Ratio / Burden resistance = 2000/120 = 16.66 
Change the calibration value from the default 111.11 to 16.66

In order to calculate Apparent Power, you multiplied Irms by 220.0V. Did you measure your mains supply and then use this value? In the second image & code, you set the voltage to 234.26V.
Set the voltage calibration values to the value of your mains AC supply read by your Multimeter.

jonathanrjpereira, 2018-04-16:
@AbrahamYu In the second image & code, you mentioned an AC-DC adapter with specifications:

> adapter
> input - AC100~240V 50/60Hz 0.2A
> output - DC9V 0.5A

But Open Energy Monitor measures the AC voltage using an AC-AC adapter & hence it requires scale down & offset circuit so that there is no negative component. 

[OEM - Voltage Sensing - AC-AC Adapter](https://learn.openenergymonitor.org/electricity-monitoring/voltage-sensing/measuring-voltage-with-an-acac-power-adapter)

You must set the voltage calibration value to the value read by the multimeter at the AC mains supply.
Adjust the phase shift value so that Real & Apparent Power are equal for a resistive load.

[Calibration](https://learn.openenergymonitor.org/electricity-monitoring/ctac/calibration)

closed 2020-01-30 by TrystanLea