openenergymonitor/EmonLib #41: How to Measure Active & Reactive Power

jonathanrjpereira, 2018-04-14:
I wanted to make a circuit that measures Active & Reactive Power drawn by AC loads. In order to do so, I had to design a circuit to measure the Phase Difference between AC voltage & current.

The idea is to use a Current Transformer(1:1000 turns ratio, 40ohm burden resistor) & a Step Down Voltage Transfromer(230V to 5V) to measure the current & voltage. The current & voltage signals would be fed to a Zero Cross Detector(ZCD) made using Comparator(LM339) & ExOR Gate(7486). The phase difference can be calculated based on the time period of the ZCD output signal.

I also used a Non Inverting Amplifier(using LM324) to amplify the input current signal to the ZCD. I have ensured that the VCC signal to all IC's is more than the Voltage of the input signals.

I am facing a few problems & not getting the intended results. I am getting proper current & voltage signals at the secondary sides of both transformers using a CRO. I am not getting the intended results at the ZCD part(both output of comparators & ExOR). i.e. The Comparator outputs are not in phase with the input voltage & current waveforms & do not show proper patterns either. The IC's tend to heat up when the voltage transformer signal is provided.

I have been working very hard to debug the circuit but to no avail. Can someone please help me out by pointing out my mistakes that I may be overlooking.

Heres a link to some of the circuit design that I have mentioned: [Circuit Design Schematics, Graphs & Formulas](https://photos.app.goo.gl/UD02mxtuHLN4z1ME2)

jonathanrjpereira, 2018-04-14:
Does EmonLib have the capability to measure Active & Reactive Power using an Arduino Uno/Nano without using a Hardware-based Zero Cross Detector Circuit?

Can I simply connect the circuit as shown in the tutorial: [OEM Mains Voltage & Current](https://learn.openenergymonitor.org/electricity-monitoring/ctac/how-to-build-an-arduino-energy-monitor) i.e. Only add the DC Bias circuit to the Current & Voltage Transformers & find the values of Active & Reactive power using EmonLib?

icboredman, 2018-04-14:
It seems that you're powering your OpAmp from a single positive supply, whereas your input signal from current transformer could swing to negative voltages. As a result, half of your signal is shorted throught an internal ESD protection diode of the LM324. Maybe that's why is it heating up.

Also, the output of the OpAmp will again be only positive, so you can not reliably compare it with LM339 using GND as a reference.

To fix it, I would recommend to connect one side of the current transformer (pin 7) not to GND, but to a half supply voltage reference, such as a resistor divider with an AC-bypass capacitor.

I hope this helps.

jonathanrjpereira, 2018-04-14:
@icboredman is it possible to completely exclude the ZCD circuit as mentioned above?

icboredman, 2018-04-15:
Absolutely! EmonLib will measure real power, apparent power and power factor without extra hardware ZCD. Check out the example: voltage_and_current.ino

jonathanrjpereira, 2018-04-15:
With respect to the Hardware ZCD circuit design, I mentioned above. I think I figured out what I was doing wrong. I'm going to try to fix it & also test out the software method using the OEM Emon Library.

The schematic that I provided will try to drive the input of the op-amp below ground for half of the cycle. The chip's internal ESD diode will clamp this, but it wasn't built for this much current. I suspect this is the source of your chip heating. 

The IC's diode clamp will prevent your VT and CT from going negative. This will saturate the transformers and cause them to have funny waveforms. I'll fix this and see if things improve.

I need to externally limit the signal so it can't try to drive the input below the Max Vin- spec. (Probably a resistor and Schottky diode clamping to ground)

closed 2018-04-15 by jonathanrjpereira