openenergymonitor/EmonLib #42: Voltage & Current Calibration

jonathanrjpereira, 2018-04-15:
When I ran the `current_only` sketch, I set the value of the current calibration to 4.75. The load I used was a 40W incandescent bulb. The real power drawn as measured by the Arduino was also ~40W.

When I ran the `voltage_and_current` sketch, I kept the same current calibration value of 4.75
I changed the value of the voltage constant to the value read from the AC mains by my multimeter.
In my case, the multimeter read 239V. Hence `emon1.voltage(2, 239, 1.7);`

For the same 40W Incandescent bulb as my load, I got the following results:
```
-70.34 71.80 430.72 0.17 -0.98 
-70.61 71.94 432.79 0.17 -0.98 
-70.51 72.16 430.68 0.17 -0.98 
-70.55 71.95 433.02 0.17 -0.98 
-69.53 70.60 432.65 0.16 -0.98 
-71.25 72.74 432.78 0.17 -0.98 
-70.85 71.95 430.73 0.17 -0.98 
-70.83 72.30 432.91 0.17 -0.98 
-69.46 71.23 432.76 0.16 -0.98 
-69.84 71.43 432.77 0.17 -0.98 
-70.23 71.53 432.81 0.17 -0.98 
-71.20 72.46 432.89 0.17 -0.98 
-70.06 71.87 430.98 0.17 -0.97 
-69.68 70.69 432.69 0.16 -0.99 
```
An incandescent bulb is considered as a Resistive Load. Hence the values of Real & Apparent power should ideally be equal.
In my case, the absolute values of Real & Apparent Power are almost equal with an error margin of ~1.4%.
The value of Irms is correct i.e. 40/239 = 0.167

I am getting a few errors that I wish to debug:

1. The value of Real Power is 70W instead of 40W. Real Power is negative whereas Apparent Power is positive.
2. The value of Vrms is 431V instead of 239 x 1.41 = 338. Is this value supposed to have an error so large?
3. The value of power factor is negative. Shouldn't this value be negative only if the load generates power which flows back to the source?

jonathanrjpereira, 2018-04-17:
Could this problem have arisen because I haven't properly calibrated the values of R1 & R2 (in my case I used a Potentiometer).

> R1 and R2 need to be chosen to give a peak-voltage-output of ~1V

From which point do I measure this peak voltage? What is the reference point; Is it Ground?  Which mode do I set my multimeter too while measuring this voltage?

jonathanrjpereira, 2018-04-18:
So here what I have understood from the forums.
The secondary winding voltage stated will be the voltage at full load. But we will need to find the voltage on no load (using a Multimeter), then recalculate the resistor values so that you get about 1 V across the 'bottom' resistor.

jxm1, 2019-09-26:
Hi im new to this, so im having the same problems. May I know what you did to solve this?
My current calibration is 54.5 as I am using a 33 ohm burden resistor.

Aldair170496, 2019-10-30:
> Hi im new to this, so im having the same problems. May I know what you did to solve this?
> My current calibration is 54.5 as I am using a 33 ohm burden resistor.

My currente calibration is also 54.5 using the same burden resistor. 
I am using a 230/9 Vrms transformer, R1=120K e R2=10K, so the voltage calibration i use is 236.45 but the voltage values is stay less than 3V, why?
Help me please!

Aldair170496, 2019-10-31:
Hi!
I fix the problem, was in the hardware not in the software.
But when the sensor or the voltage transformer are disconnected it should be all zero (0), how to fix that.
Any help would be helpfull.

jonathanrjpereira, 2019-11-01:
@jxm1 I haven't managed to fix the problem.

TrystanLea, 2020-01-30:
Hello @jonathanrjpereira the negative values reported above sound like the result from having the CT sensor connected the wrong way around, the values should be correct if you flip the sensor 180 degrees.

Could your voltage calibration also be incorrect?

closed 2020-05-21 by jonathanrjpereira