openenergymonitor/EmonLib #53: CalcVI async calculation

felangga, 2019-10-17:
Is it possible to make calcVI function do asynchronous calculation? 
Because I cannot control other switch while emonlib doing calculate

Aldair170496, 2019-10-30:
Hii! My objective is to read P, S, FP, Vrms and Irms.
My currente calibration is 54.5 using a 33 burden resistor.
I am using a 230/9 Vrms transformer, R1=120K e R2=10K, so the voltage calibration i use is 234.26 but the voltage values is stay less than 3V, why?
Help me please!

CODE:
{  
  Serial.begin(9600);
  
  emon1.voltage(2, 234.26, 1.7);  // Voltage: input pin, calibration, phase_shift
  emon1.current(1, 54.5);       // Current: input pin, calibration (2000/33).
}

void loop()
{
  emon1.calcVI(20,2000);         // Calculate all. No.of half wavelengths (crossings), time-out
  emon1.serialprint();           // Print out all variables (realpower, apparent power, Vrms, Irms, power factor)
  
  float realPower       = emon1.realPower;        //extract Real Power into variable
  float apparentPower   = emon1.apparentPower;    //extract Apparent Power into variable
  float powerFActor     = emon1.powerFactor;      //extract Power Factor into Variable
  float supplyVoltage   = emon1.Vrms;             //extract Vrms into Variable
  float Irms            = emon1.Irms;             //extract Irms into Variable
  delay(1000);
}

Aldair170496, 2019-10-30:
sensor de corrente SCT-013-000

felangga, 2019-11-01:
@Aldair170496 dude please open new thread

TrystanLea, 2020-01-23:
>Is it possible to make calcVI function do asynchronous calculation?

Not that I'm aware I'm afraid. You can reduce the time taken and space out the different channels if you need to check or control a button at other times. Or maybe it is easier to dedicate a second micro for the control part?

asahajit, 2020-11-01:
If You are using esp32 You can try Multithreading as it has 2 core
Energy Calculation in one core0 and switch control on core1

```
xTaskCreatePinnedToCore(
      Task1code, /* Function to implement the task */
      "Task1", /* Name of the task */
      10000,  /* Stack size in words */
      NULL,  /* Task input parameter */
      0,  /* Priority of the task */
      &Task1,  /* Task handle. */
      0); /* Core where the task should run */

Void Task1code( void * parameter) {
  Your Emonlib Calculation 
  }
```
}