openenergymonitor/EmonLib #56: Arduino Wifi rev 2 compatibility

1omarsaid, 2020-02-27:
It doesnt seem like this library works with the arduino uno wifi rev 2 due to the change of the analog pin names.  I get the following error:

`Arduino: 1.8.13 Hourly Build 2020/02/19 04:12 (Mac OS X), Board: "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp: In member function 'long int EnergyMonitor::readVcc()':
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:251:3: error: 'ADCSRA' was not declared in this scope
   ADCSRA |= _BV(ADSC);                             // Convert
   ^~~~~~
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:251:3: note: suggested alternative: 'ADC_t'
   ADCSRA |= _BV(ADSC);                             // Convert
   ^~~~~~
   ADC_t
In file included from /Users/omarsaid/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/io.h:99:0,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/pgmspace.h:90,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/String.h:30,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/Print.h:24,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/Stream.h:25,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/Client.h:22,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/omarsaid/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/Arduino.h:23,
                 from /Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.h:15,
                 from /Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:13:
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:251:17: error: 'ADSC' was not declared in this scope
   ADCSRA |= _BV(ADSC);                             // Convert
                 ^
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:251:17: note: suggested alternative: 'ADC0'
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:253:12: error: 'ADCL' was not declared in this scope
   result = ADCL;
            ^~~~
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:253:12: note: suggested alternative: 'ADC0'
   result = ADCL;
            ^~~~
            ADC0
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:254:13: error: 'ADCH' was not declared in this scope
   result |= ADCH<<8;
             ^~~~
/Users/omarsaid/Documents/Arduino/libraries/EmonLib-master/EmonLib.cpp:254:13: note: suggested alternative: 'ADC0'
   result |= ADCH<<8;
             ^~~~
             ADC0
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

Any way I can change the code to make it compatible with the arduino uno wifi rev 2? I know I need to make some changes to the "ADC" and looking online, I see that they changed the names to AIN but I am not sure exactly what to do.

TrystanLea, 2020-03-16:
Hello @1omarsaid we are not targeting support for Arduino Wifi so Im afraid it is not likely that this will be fixed. Let us know if you work out what is needed to fix

J0J015JK, 2021-10-05:
I changed the registers so that Emonlib will work on the Arduino UNO WiFi
I called it EmonDIY so just change the #include Emonlib line to include EmonDIY
Download link => https://github.com/J0J015JK/EmonDIY

gbtassara, 2021-11-22:
Dear J0J015JK, it seems your lib does not work on Arduino UNO WIFI rev2...
it seems that MUXPOS, REFSEL0, AIN1, AIN2, AIN3, ADCn.COMMAND, STCONV, ADCn.RESL, ADCn.RESH are not defined...