openenergymonitor/EmonLib #60: Wrong (lower) current values with SCT-013-030?

dotto59, 2020-06-09:
I have an SCT-013-030 (30A, 1V peak signal, internal burden resistor 62 Ohm) connected to a UNO as "regular" schema (voltage divider with 2x10k from +5V to GND, and a 100nF condenser to GND). 

My target is to monitor my Photovoltaic production, the same way I did before for years with an Efergy e2 power monitor (you can see it on the enclosed picture) that uses its own similar clip current sensor but without sensor specs I opted for that SCT-013. I'm aware that the measure can't be exact, as can vary depending on the load characteristics, but I can say Efergy always gave me not more than +/-5% difference from the inverter readings, so I expected the same range with SCT, Emonlib and Arduino UNO. 

So, I made this simple sketch to perform some tests and calibration:
`#include "EmonLib.h"
EnergyMonitor emon1;
//Pin del sensore SCT su A1
#define SENS_CORRENTE A1
void setup()
{
  Serial.begin(115200);   //Apro la comunicazione seriale
  Serial.print("Calibrating...");
  // Just a bit of void readings to stabilize...
  for (int i=1; i<=4; ++i)
    analogRead(SENS_CORRENTE);
  delay(500);
  // Then calibrate (const= Ratio/Res. Burden 1800/62 = 29).
  emon1.current(SENS_CORRENTE, 29);
  Serial.println();
}
  
void loop()
{
  // Read current
  double Irms = emon1.calcIrms(1480);
  // Just to ignore small fluctuations
  if ( Irms <= 0.13 ) Irms = 0;
  // Power ad current x 230V line voltage
  int watt = (Irms*230);
  watt = watt * 10;
  // Show values
  Serial.print(Irms,2);
  Serial.print(" ");
  Serial.println(watt);
  delay(1000);
}`
The problem is I always get a power reading LOWER than real. For example, if I clip the sensor over the active wire of a power cable and connect a plug-in energy monitor and then a hairdryer, the energy monitor gives me the correct 1.7kW value, but that Arduino sketch gives me around 950W!
I then connected the sensor to the PV line out (the same wire I used with Efergy...) and here again I must multiply the resulting power by 1.7 to have a "similar" reading than the PV inverter says, but this 1.7 factor seems to be somehow "variable", depending on the PV output power, sometimes lower and sometimes higher for a bunch of hundred of watts, with an estimated factor ranging from 1.6 to 2.1 to get almost "exact" values!

I can accept a +/-5% precision like Efergy did, even a bit more, but an almost constant +70% is unacceptable. It seems something is wrong either with my circuit (but it's pretty simple..), and/or my implementation (but the measure sketch is like the example one) and/or the sensor (could be defective?), and/or the library (but I don't thik so), and/or what?
Thanks!
[image]

bottosac, 2020-10-27:
I'm in your same situation. Need to replace Efergy with Arduino energy monitor for my PV.
I'm more near real values but with a burden resistor of 75ohm and a calibration value of 180. But in some ranges I'm away from real value of 200-300W

beckmx, 2021-08-22:
@dotto59 how did you get to this value?
`Then calibrate (const= Ratio/Res. Burden 1800/62 = 29).`
1800 I suppose is the average value at 0 volts consumption?

wangeris, 2023-08-21:
did you ever figure it out?