openenergymonitor/EmonLib #72: Solve most of your issues, ESP32

alexmoz2016, 2023-04-02:
Some code that you may want to change to make it work (I have ESP32):
In EmonLib.h:
```
	#if defined(__arm__)
	#define ADC_BITS    12
	#else
	#define ADC_BITS    10
	#endif
```
Just add/replace with:
`#define ADC_BITS    12`

Because ESP32 is a 12-bit ADC.

In EmonLib.cpp
Replace all occurences of "/1024" by "/(ADC_COUNTS>>1".
The code was originally written for a 10-bit ADC, without ability to change it to 12 bits.

That's all. The precision and predictability of the reading is improved a lot.

Another note, for Calibration (111.1 value). If you have a voltage-output sensor (SCT-013-015, SCT-013-020, etc.), replace the "111.1" with "15" for -015, "20" for -020, etc. In other words, use the same calibration value as the Amp/V value (15A/V use "15", 30A/V use "30"). Most of the examples I saw online, using the 111.1 calibration value, are only good for SCT-013-000 with current output (which also require a burden/load resistor). Since the non-000 (-015, -020, -050, etc.) sensors have the burden/load resistors integrated, the calibration constant is different.

I hope it helps someone. Took me a few days to figure it out.

Alex

alkaraschu, 2023-06-04:
"Replace all occurences of "/1024" by "/(ADC_COUNTS>>1"."
Probably a bracket missing?

DieWaldfee, 2023-07-15:
"Replace all occurences of "/1024" by "/(ADC_COUNTS>>1"."  --> Error cause of missing ")"
should be:  "/1024" --> "/(ADC_COUNTS>>1)"
(as @alkaraschu mentioned)

staadecker, 2023-12-14:
If I'm not mistaken, you should really change all values of `1024` to `ADC_COUNTS` not `(ADC_COUNTS >> 1)`. The former is equal to `4096` and the latter `2048`. If you want to match the behavior of the original `1024` code, then `4096` is the current value for a 12-bit device.

It doesn't matter too much though since these values are just for the low-pass filter and either `4096` or `2048` are likely large enough to work.

peteDDD, 2024-02-04:
Note that the ESP32-S2 has a 13-bit ADC...