openenergymonitor/EmonLib #75: High reading at the beginning

hayderismael, 2023-09-26:
I have a problem measuring the Amps and kWhs for a few devices. The lib is so precise; nevertheless,  after a while. In the beginning, I received a very high reading, and it kept reduced for a few cycles of the code until it settled to the precise value.  Please advise. 

Thank you.

peteDDD, 2024-02-04:
I am experiencing the same issue.  Here is an example of my data for Vrms on a 120VAC 60hz line.  My calls are  
```
emon1.voltage(VOLTAGE_PIN, CALIBRATION_VOLTAGE, 1.7)
emon1.calcVI(30,2000)
```

and here is my data, collected using double-buffering (into two buffers) and freeRTOS tasks, one for collecting the data and one for formatting and transmitting it.  Each buffer collects 60 samples of 30 zero-crossings.  The data really is "all over the place".  Suggestions?

jsonPayload = [202.36,207.04,204.42,204.58,204.11,197.25,196.75,195.85,196.28,202.12,194.21,199.55,196.19,196.18,192.40,196.45,192.51,189.70,187.56,192.37,192.68,197.98,192.90,192.92,188.10,191.15,191.23,192.68,190.68,187.25,182.89,187.23,179.76,179.20,184.23,185.53,186.32,190.52,181.94,179.35,180.94,175.87,185.37,176.22,187.83,178.59,179.37,177.65,169.26,171.55,184.82,175.99,177.49,176.13,179.60,175.19,175.88,175.01,176.00,178.54]
jsonPayload = [178.95,178.73,167.17,174.13,169.15,167.00,174.95,168.27,171.48,172.97,172.02,166.65,169.44,163.00,165.33,174.17,157.81,160.53,165.53,164.59,171.21,162.92,161.55,161.96,166.52,164.31,166.81,160.37,155.86,158.92,162.05,160.01,161.94,161.75,160.87,157.65,155.46,164.46,157.27,161.30,146.14,166.31,149.56,156.16,155.73,149.56,154.96,151.46,150.04,146.32,148.66,156.74,151.79,144.44,157.80,150.40,151.88,157.49,152.02,151.68]
jsonPayload = [146.02,147.17,142.78,142.74,140.22,143.21,152.82,146.29,140.65,146.33,142.48,148.87,144.78,144.33,147.51,150.55,148.85,144.35,143.15,148.61,139.87,142.09,146.45,143.20,154.86,141.16,139.90,142.43,138.82,150.43,141.49,142.40,145.36,147.64,143.99,145.77,145.47,142.04,136.57,134.35,139.22,134.03,143.82,140.02,138.39,142.15,143.30,143.84,135.95,141.61,139.65,139.01,141.85,142.53,140.13,140.76,142.70,137.92,141.00,134.13]
jsonPayload = [135.76,138.38,136.53,133.57,139.09,137.12,133.53,131.51,134.40,135.74,132.56,133.50,135.83,131.14,131.41,141.12,132.41,129.55,129.30,137.33,132.29,137.85,137.17,132.49,128.87,133.46,130.94,134.24,138.21,130.14,132.28,132.38,130.79,130.34,130.55,129.52,133.08,136.06,134.96,134.03,137.98,130.93,132.58,127.52,126.12,131.50,133.73,132.63,124.59,124.71,125.15,131.38,131.87,135.70,135.85,131.78,127.81,130.53,128.30,128.41]
jsonPayload = [125.78,131.92,125.57,126.13,132.20,133.93,124.51,133.90,136.84,126.44,132.61,129.57,127.82,130.01,134.26,132.23,125.34,129.54,125.18,131.73,131.89,124.27,124.88,124.75,123.71,130.66,130.50,125.12,129.13,126.07,124.70,127.34,129.44,127.68,126.73,126.25,122.39,126.75,120.17,126.87,132.80,130.39,132.67,123.35,124.90,120.86,124.03,128.26,129.48,124.02,126.76,124.59,124.01,124.05,126.58,127.88,122.15,125.74,127.04,125.15]
jsonPayload = [129.77,131.52,133.33,127.67,122.44,120.13,125.60,119.21,127.80,124.19,119.35,122.83,125.91,127.79,123.27,117.16,127.55,127.39,126.72,127.04,122.38,123.16,123.01,121.02,126.77,121.82,119.30,130.32,127.23,126.80,123.78,124.89,120.18,124.80,125.98,124.94,122.53,125.66,125.87,124.80,122.42,121.85,128.73,118.43,123.98,121.23,124.47,121.57,123.64,119.34,120.51,120.23,119.17,128.06,125.78,127.24,125.95,118.55,116.31,122.76]
jsonPayload = [119.10,127.64,124.51,120.73,116.40,118.65,125.03,121.32,116.83,120.18,125.48,125.34,125.97,118.28,122.83,119.66,118.47,121.30,119.16,123.20,125.97,123.59,124.58,116.35,119.83,121.81,126.24,121.15,118.89,117.36,116.08,123.69,126.95,119.06,122.84,124.02,124.73,118.98,122.46,120.00,116.72,117.82,127.88,115.69,121.91,118.17,117.57,117.55,125.45,120.12,122.89,126.17,118.92,123.37,123.29,117.34,120.68,118.33,119.60,118.54]
jsonPayload = [118.51,123.91,124.47,121.30,117.94,118.33,118.46,117.05,122.31,118.26,125.32,123.92,119.54,124.10,120.71,115.72,116.85,116.97,123.60,118.27,121.43,119.87,115.31,117.03,115.29,120.94,121.32,119.21,116.50,122.12,124.05,125.48,125.93,125.06,124.68,122.56,120.24,124.39,123.19,120.22,121.51,119.36,116.44,115.28,116.64,120.56,116.62,118.67,121.95,125.64,120.13,120.76,120.18,121.02,118.82,116.61,124.95,121.73,122.05,118.95]
jsonPayload = [123.81,119.46,122.24,124.01,123.70,118.15,118.36,118.39,119.53,122.15,118.77,118.45,116.08,116.59,121.01,119.65,120.99,122.09,124.64,123.36,122.37,119.16,120.77,117.60,120.56,118.78,114.38,123.12,119.04,121.37,126.02,124.63,122.07,123.17,124.63,115.21,118.18,118.40,117.25,122.07,121.50,119.88,121.45,119.01,120.81,120.85,121.13,117.76,117.79,126.42,124.14,125.36,121.32,122.45,121.87,122.10,118.83,119.05,118.05,121.83]
jsonPayload = [125.73,117.25,121.30,117.76,115.67,117.10,117.88,115.98,119.83,114.77,122.81,121.84,120.50,114.31,119.92,117.67,118.99,115.24,123.27,123.92,122.35,118.50,122.63,123.41,121.71,116.10,120.79,123.99,116.59,116.37,120.21,119.42,116.29,120.68,124.15,123.29,118.64,114.34,118.54,118.70,116.95,119.96,122.60,121.65,120.47,123.89,116.61,117.19,124.67,122.91,117.51,123.53,120.79,120.59,120.18,121.66,113.21,122.79,117.78,119.02]
jsonPayload = [124.94,123.38,116.92,117.37,118.89,119.69,119.07,116.45,117.92,120.37,126.97,117.13,117.85,119.52,119.96,119.71,122.99,117.93,122.94,116.62,121.20,121.20,122.46,120.27,123.44,117.72,115.20,117.26,120.24,117.38,116.15,121.22,116.24,118.14,116.82,115.45,124.40,125.32,120.45,114.85,119.46,119.56,120.34,119.15,116.29,122.72,121.94,123.40,122.82,115.33,117.37,114.18,114.50,122.13,122.09,116.49,115.02,119.22,122.30,117.98]
jsonPayload = [125.82,117.18,116.43,116.46,115.13,119.73,122.43,122.75,116.38,124.37,124.00,117.03,117.95,115.05,118.40,119.46,122.27,114.62,117.41,116.81,115.33,119.06,119.79,117.15,122.72,118.53,118.24,116.47,118.94,123.41,120.8 [... cut ...] ,121.71,121.35,120.49,112.28,116.71,116.53,120.58,121.22,118.09]
jsonPayload = [121.47,123.64,119.01,116.50,118.07,117.37,117.29,115.04,114.45,117.66,119.03,118.44,124.92,120.59,118.39,122.35,118.57,122.03,122.47,116.93,115.91,115.76,117.10,114.70,116.71,119.34,123.96,125.89,117.13,118.36,120.60,122.78,122.11,122.66,122.64,117.27,115.82,121.06,121.82,122.77,123.00,123.21,116.07,118.39,119.93,121.79,119.78,122.90,116.51,116.54,121.55,114.04,116.03,116.56,116.49,121.64,117.13,115.69,116.33,114.46]

peteDDD, 2024-02-04:
And for those who prefer a graphical view

[image]

peteDDD, 2024-02-04:
And, I'm pretty sure my power line is not THAT noisy.

hayderismael, 2024-02-04:
ّI have precisely the same form shape when I start running my system. It might say that it is just for a few seconds, but I have a big load, and it adds few extra GBPs £ for consuming nothing.

peteDDD, 2024-02-05:
I dug into this a bit more and now understand why the high readings at first...
It has to do with the calculation in the library which tries to find the vOffset for each reading.  It starts with an assumption that the crossing voltage is at exactly half one half of the ADC range: 
`offsetV = ADC_COUNTS>>1;`
That's fine, if it were correct.  In my case though, the zero-crossing ADC count is about 5080 (with a 13-bit ADC that has a top count of 8192).  So it is more like 62% of the ADC range.
The code then does a "low-pass filter" to get closer to the zero-crossing offset point with each voltage reading:
```
    //-----------------------------------------------------------------------------
    // B) Apply digital low pass filters to extract the 2.5 V or 1.65 V dc offset,
    //     then subtract this - signal is now centered on 0 counts.
    //-----------------------------------------------------------------------------
    offsetV = offsetV + ((sampleV-offsetV)/ADC_COUNTS);
    filteredV = sampleV - offsetV;
```
So, overtime, it does in fact get closer but I believe it still tends to "drift" that center point.

So I changed the initial offset to my experimentally (on the bench with oscilloscope) zero-crossing count.
```
 #define OFFSET_COUNT 5080 // experimentally found the crossing point using 3.3 V supply
// (I actually maintain all of these defines in a separate settings.h file)

//offsetV = ADC_COUNTS>>1;
  offsetV = OFFSET_COUNT;  // PD
```
[image]

Now, we start off in about the right place, but the noise is still very high...

I am trying some alternatives and will report back.

razarahil, 2024-11-22:
> I dug into this a bit more and now understand why the high readings at first... It has to do with the calculation in the library which tries to find the vOffset for each reading. It starts with an [...]

Did you fix the issue and manage to get stable reading?